riscv/riscv-spmp #17: Atomic SPMP enable/disable on RV32 platform

reosdev, 2024-11-21:
I'm developing the real-time OS for RV32 platform with SPMP. OS design assumes the run-time PMP configuration update (every task has own memory regions that must be isolated from OS Kernel and other tasks). I do see the problem with SPMP enable/disable which cannot be performed atomically, as there are two `pmpswitch` registers that need to be written. Operation in between of these writes is unsafe as the accessible memory map is inconsistent with only half of SPMP configuration. Is there is an alternative way to atomically disable/enable the SPMP?

ybc-alkaid, 2025-03-25:
Hi reosdev,
Sorry for the late response.
To be honest, I don't see the problem here.

I think an implementation should implement spmpswitch as two masks under RV32 (i.e., if you modify spmpswitch0, the status of spmp0-spmp31 should change accordingly).
This is similar to directly modifying multiple SPMP CSRs.
So, two CSR writes between spmpswitch0/1 shouldn't cause inconsistent memory accesses.

reosdev, 2025-03-28:
Hello ybc-alkaid,

Thanks for responding. Seems like my description was not clear enough. Lets try once again and consider the simplest scenario - SPMP disable/entire_configuration_update/enable. There is the sequence:

	1. spmpswitch0 = 0
	>> excution here is NOT SAFE!!!
	2. spmpswitch1 = 0
	3. load spmpaddr{i}/spmpcfg{i} from some memory location
	4. spmpswitch0 = 0xFFFFFFFF
	>> execution here is NOT SAFE!!!
	5. spmpswitch1 = 0xFFFFFFFF
	
Due to the prioritization between entries we cannot guaranty the consistency of the protected memory map when only half of PMP entries are enabled (in between of 1 and 2, 4 and 5). To make the SPMP reconfiguration procedure be safe, the enable/disable must be atomic.

ybc-alkaid, 2025-04-19:
I get you point that code executed between switches may cause inconsistency.
However, even without using spmpswitch, the same issue exists when updating SPMP registers, e.g., spmpaddr0 and spmpcfg0 (the same applies when configuring PMP registers). 

From a software perspective, S-mode software must ensure the code configuring SPMP is atomic.
From a hardware perspective, there are two possible scenarios in which instructions can occur between SPMP re-configs.
One is when an NMI occurs during SPMP configuration, in such a case, the handler should make sure it won't violate the SPMP rules (depending on implementation, it can be achieved by switching the SPMP to a dedicated configuration).
Another possibility is transient execution. In such a case, it is implementation-specific whether to track the dependency between memory access and SPMP configuration instructions. But there still may exist side-channel windows.

Additionally, the SPMP is per-hart, therefore current hart's reconfiguration will neither affect nor be affected by other harts.