riscv/riscv-spmp #25: confused statements about spmpcfg.SRWX=1111 between 0.9.1 and 0.9.2

up2u, 2025-03-25:
Hi maintainer,

some confused statements
from 0.9.2 pdf (commit : 328e287222500dfe7974dd8e9713f1d446814bd8)
[image]

there is:
" The encoding spmpcfg.SRWX=1111 is used for backward compatibility, where both S-mode and
U-mode have RWX permissions."

but from 0.9.1 pdf
[image]

it is 
"The encoding spmpcfg.SRWX=1111 can be used for sharing data between S-mode and U-mode,
where both modes only have read-only permission to the region."

There are 2 questions from me
1: spmpcfg.SRWX=1111 in 0.9.1 its really permission is shr RO while 0.9.2 is enforce + enforce + deny ?  so 0.9.2 is not backward compatibility ?
2: if it's not backward compatibility, so in in 0.9.2, its really permission is enforce + enforce + deny?
2.1 if it's backward compatibility, its permission is shr RO ?

ybc-alkaid, 2025-03-25:
Thanks for your comments, you've found the blind spot.
That is a legacy statement, sorry for the confusion.
Back in version 0.9.2, "spmpcfg.SRWX=1111" was encoded as "Allowed", giving full permission to S/U.
Such encoding can be seen as a way to disable the SPMP (set the first rule as 1111 with a full address range).
But then we realized it is more reasonable to keep it as enforce + enforce + deny.
And if needed, the implementation can use the RSVD encoding to disable SPMP.

I will remove this sentence, thanks for pointing it out!

up2u, 2025-03-26:
Thanks for your reply.

one more question, from your statement:
"
Back in version 0.9.2, "spmpcfg.SRWX=1111" was encoded as "Allowed", giving full permission to S/U.
Such encoding can be seen as a way to disable the SPMP (set the first rule as 1111 with a full address range).
"

so  is there a history version of 0.9.2 ( i see 0.9.2 pdf file with Version 0.9.2, 5/2024 @20250317,  difficult to distinguish the "in development" version ) with entry : spmpcfg.SRWX=1111 that "Allowed", giving full permission to S/U.
and meanwhile the entry was used as disable SPMP ? 

my point is "full permission to S/U" vs "disable SPMP"  that violate intuition, too difficult to accept such motivation.
this may be history story(no effect the current design, just follow the latest spec),  query this  just for more understanding。

Thanks a lot

ybc-alkaid, 2025-03-26:
The current version should be 0.9.5 (we forget to update the version number for 0.9.3/4).
The reason "full permission to S/U" equals "disable SPMP" is that the spec states:
> If the effective privilege mode of the access is U and no SPMP entry matches, but at least one SPMP entry is implemented, the access is denied;

Simply setting spmpcfg.A=0 for all entries can not disable SPMP for U-mode.
That's the story behind the strange encoding of "full permission to S/U".

closed 2025-04-10 by ybc-alkaid