riscv/riscv-spmp #46: Should we allow more PMP_Resource entities in the pool?

bgrayson-mips, 2025-08-10:
The spec as written allows at most 64 total PMP_Resource entities to be shared among PMPs and SPMPs. Given that we can have up to 64 PMPs and up to 64 SPMPs, it is not unreasonable to allow the pool to be larger than 64. For example, with 96 PMP_Resource entities, one could have 64 PMPs and 32 SPMPs, or 48 PMPs and 48 SPMPs, both of which fit within the existing architectural limits for both types. Given the stated preference for using TOR-style addressing for SPMPs and thus requiring more SPMPs than would otherwise be needed via NAPOT addressing, extending the PMP_Resource to up to 128 entities, with additional rules that neither PMP nor SPMP resources could be more than 64 each, seems reasonable. (Any upper entities beyond the limits would be unusable.)

ybc-alkaid, 2025-08-16:
We acknowledge the desire to provide more resources/entries. However, SPMP is using a SINGLE PMP_Entries pool shared with PMP, as requested by the ARC. As PMP currently allows only 64 entries, we are limited here. A reasonable, still clean way out, will be to define a further sub-extension, which extends the PMP_Entries beyond 64, and also allowing configs like 24, 32, 48. The working title of the extension can be Mpmpflex.

allenjbaum, 2025-08-17:
I don't think that the requirement to share the register pool with PMP should be a limiter.
It could still allow only 64 entries to be shared, yet allow additional unshared entries.
e.g. NumPMP would still be limited to be 0..64.
Discovery (of how many additonal entries there are) , and reserving enough siselect encoding space are likely  the only tricky parts. Right now there must be either 16 or 64 entries in total (I'd argue it can't be zero if SPMP is implemented at all...).

ybc-alkaid, 2025-08-22:
> E.g., say we have 80 PMP_entries in total, 64 of which are delegated to SPMP. Then how do you access SPMP 0..64 via siselect? Use siselect value 0..64 or 16..79?

In this case, whether we can allocate more than 64 siselect values determines the physical and logical addressing.

Unfortunately, since the maximum number of SPMP entries is 64, it is unlikely to allocate more than 64 siselect values for SPMP.
Thus, the indirect access of SPMP must be handled logically. Otherwise how can SPMP[64], which is practically PMP_entry[79], be handled physically?

allenjbaum, 2025-08-24:
I thought he whole point of this exercise is to expand the number of SPMP
entries to be>64.
If you are limiting it to 64 SPMP entries + N PMP entries, you could
possibly save an siselect bit,
but a single select bit is cheaper than all the surrounding logic it would
take to make it work logically rather than physically

If AR won't allow more than 64 entries, this is all moot. Change nothing
If AR will allow more, then there will must be more than 64 legal
siselect values, and they can be
SPMPBase+0..SPMPBase63, SPMPBase64..SPMPBaseN,
     or they could be
SPMPBase+0..SPMPBase+63, SPMPBase2+0..SPMPBase2+N
     where SPMPBase could be a value that is not contiguous
    ( but ideally should be SPMPBase+2^S for some S>6

allenjbaum, 2025-08-24:
> Exactly, the ARC doesn't want to add more PMP entries than 64 at this
> point.
> Otherwise, there's not much to debate, simply adding more siselect values
> will do.

Then we don't need to change the existing spec, correct? What am  I
missing? What are we debating?

closed 2025-10-15 by ybc-alkaid